learnsyslab/gym-pybullet-drones #121: Multi-agent leaderfollower clarification

camillo2008, 2022-10-28:
Hello, 
I am noticing that when I train multiagent with leader-follower, sometimes in one dimension control the drone start to fly around in 3D space, leaving the Z axis, to me, it seems that this appends when one of the two drones touch the floor. 
Does the collision have something to do with this?

This is really noticeable, with gui=True, do you have any idea why this is happening?

And another thing, now the reward function in leader follower seems to be intended for more than one follower drone, like 1 leader 10 follower, 
I have read the issue:https://github.com/utiasDSL/gym-pybullet-drones/pull/33, still i don't get if multiagent can contain more than 2 drones, the other just falls to the floor.

JacopoPan, 2022-11-03:
W.r.t. the 1D action spaces leading to 3D movement, it almost certainly relates to collisions with objects in the environment.

W.r.t. using more than 2 agents: it is supported on the environment side but not by the RL agent in `multiagent.py`, even if you add multiple policies, you should re-write the centralized critic, the observer space and how the centralized critic observations are filled.

camillo2008, 2022-11-12:
I understand thank you for the reply

closed 2022-11-12 by camillo2008