learnsyslab/gym-pybullet-drones #122: how can I optimize PID coefficients by TuneAviary.py

powehi2020, 2022-11-02:
hi~ if there are an example to use it？or it should DIY？I have not find the example in readme？

JacopoPan, 2022-11-03:
TuneAviary is an aviary with a 6-dimensional action space that is used to adjust the default PID coefficients on a fixed trajectory. It is accessible through `singleagent.py` like any other single agent aviary.

powehi2020, 2022-11-03:
thank you for you answer!  can you give me some advice to realize PID coefficients on downwash.py in the lower drone? should i rebuild an env in BaseMultiagentAviary.py?

JacopoPan, 2022-11-03:
I would probably just start from `downwash.py`, loop it over a number of training episodes N and add a custom agent/learning module that adjusts the PID coefficients of one of the controllers in the main script.

powehi2020, 2022-11-04:
> It looks like a single agent problem and I don't know if you necessarily want to use RL or some other learning/optimization approach. I would probably just start from `downwash.py`, loop it over a number of training episodes N and add a custom agent/learning module that adjusts the PID coefficients of one of the controllers in the main script.

Thank you for your answer!