learnsyslab/gym-pybullet-drones #123: Memory leak during training

4ku, 2022-11-03:
During training, my RAM is constantly decreasing. I tried to use the garbage collector (`gc.collect()`) after resetting the simulation, but it helps a little.

JacopoPan, 2022-11-03:
Can you provide additional detail (executed commands, terminal output, computer, os, python configurations)?

4ku, 2022-11-04:
I create environment in Ubuntu 20.04 according to this instruction:
```
$ conda create -n drones python=3.8 # or 3.9 on Apple Silicon, see the comment on grpc above
$ conda activate drones
$ pip3 install --upgrade pip
$ git clone https://github.com/utiasDSL/gym-pybullet-drones.git
$ cd gym-pybullet-drones/
$ pip3 install -e .
```
But reinstall PyTorch, because there is an error on the current version
```
pip install torch==1.9.0+cu111 torchvision==0.10.0+cu111 torchaudio==0.9.0 -f https://download.pytorch.org/whl/torch_stable.html
```
I've launched on two different machines with rtx 3050 ti laptop and rtx 3090 and have the same problem.
I just run modification of `learn.py` in the `examples` folder and I use vectorized environment here: `env = make_vec_env("hover-aviary-v0", n_envs=8)`

JacopoPan, 2022-11-04:
I see. Can we try to go one step at a time? What is the PyTorch installation error?

4ku, 2022-11-05:
I had this error for PyTorch 1.12.1:
```
/home/ivan/miniconda3/lib/python3.8/site-packages/torch/cuda/__init__.py:146: UserWarning: 
NVIDIA GeForce RTX 3050 Ti Laptop GPU with CUDA capability sm_86 is not compatible with the current PyTorch installation.
The current PyTorch install supports CUDA capabilities sm_37 sm_50 sm_60 sm_70.
If you want to use the NVIDIA GeForce RTX 3050 Ti Laptop GPU GPU with PyTorch, please check the instructions at https://pytorch.org/get-started/locally/
```
But now I update PyTorch to 1.13.0 and everything is ok. Before PyTorch 1.13.0 release I downgrade it to 1.9.0

4ku, 2022-11-05:
I figured out the memory leak issue. I uncommented [this line](https://github.com/utiasDSL/gym-pybullet-drones/blob/fab619b119e7deb6079a292a04be04d37249d08c/gym_pybullet_drones/envs/BaseAviary.py#L203) as recommended in the README. But if you don't touch it, then everything is ok.

JacopoPan, 2022-11-06:
Ok, thank you for looking into this, I can't reproduce the problem atm because I'm not on Ubuntu but I will remove the suggestion from the README.

closed 2022-11-06 by JacopoPan