learnsyslab/gym-pybullet-drones #126: LeaderFollower, leader only has height axis freedom.

jackvice, 2022-11-06:
Hi, I am playing with leader behavior and I can get it to hover at different heights but it does't want to leave 0, 0 for x and y? I'm guessing it may have been x,y fixed for faster demo training? I'm changing reward in line 84 of LeaderFollowerAviary.py to get new behaviors. e.g.
rewards[0] = -1 * np.linalg.norm(np.array([0, .25, 0.5]) - states[0, 0:3])**2

JacopoPan, 2022-11-07:
Can you tell me what scripts/commands/ you are running? The default examples are, as you say, going to use the "1-D" action space (i.e., forcing the force control on all propellers to be the same) that greatly simplify/accelerates the solution of the problem.

jackvice, 2022-11-07:
I made this change in LeaderFollowerAviary.py line 84:
rewards[0] = -1 * np.linalg.norm(np.array([0, .25, 0.5]) - states[0, 0:3])**2
and then ran gym-pybullet-drones/experiments/learning/multiagent.py
I thought I tried setting ACT to ActionType.RPM.  I'll have to double check.  Do [ActionType.RPM, ActionType.DYN, ActionType.VEL] all allow 3D flight?  And thank you!

JacopoPan, 2022-11-07:
That would be the intended behaviour, double check that your code is going through lines 197-198 of `BaseMultiAgentAviary.py`:
```
if self.ACT_TYPE == ActionType.RPM: 
    rpm[int(k),:] = np.array(self.HOVER_RPM * (1+0.05*v))
```
and 174-175 of `multiagency.py`
```
elif ARGS.act in [ActionType.RPM, ActionType.DYN, ActionType.VEL]:
    ACTION_VEC_SIZE = 4
```
These should lead to different actions on different propellers and movement in the 3D space

jackvice, 2022-11-09:
Setting actiontype to rpm and "timesteps_total": 500000 and both drones crashed when I ran policy using the test_multiagent.py so I set  "timesteps_total": 5000000 and got the following error running on CPU or GPU:
2022-11-09 15:50:29,427	ERROR trial_runner.py:958 -- Trial PPO_this-aviary-v0_924f1_00000: Error processing event.
Traceback (most recent call last):
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/trial_runner.py", line 924, in _process_trial
    results = self.trial_executor.fetch_result(trial)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/ray_trial_executor.py", line 787, in fetch_result
    result = ray.get(trial_future[0], timeout=DEFAULT_GET_TIMEOUT)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/_private/client_mode_hook.py", line 105, in wrapper
    return func(*args, **kwargs)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/worker.py", line 1713, in get
    raise value.as_instanceof_cause()
ray.exceptions.RayTaskError(ValueError): ray::PPO.train() (pid=926055, ip=192.168.1.110, repr=PPO)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/agents/ppo/ppo_torch_policy.py", line 82, in loss
    curr_action_dist = dist_class(logits, model)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/models/torch/torch_action_dist.py", line 185, in __init__
    self.dist = torch.distributions.normal.Normal(mean, torch.exp(log_std))
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/torch/distributions/normal.py", line 50, in __init__
    super(Normal, self).__init__(batch_shape, validate_args=validate_args)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/torch/distributions/distribution.py", line 55, in __init__
    raise ValueError(
ValueError: Expected parameter loc (Tensor of shape (128, 4)) of distribution Normal(loc: torch.Size([128, 4]), scale: torch.Size([128, 4])) to satisfy the constraint Real(), but found invalid values:
tensor([[nan, nan, nan, nan],
        [nan, nan, nan, nan],
        [nan, nan, nan, nan],
        [nan, nan, nan, nan],
        [nan, nan, nan, nan],
        [nan, nan, nan, nan],
        [nan, nan, nan, nan],
        [nan, nan, nan, nan],
        [nan, nan, nan, nan],
...

JacopoPan, 2022-11-09:
@jackvice it is easier to read terminal output if you put it in a code block with some formatting highlighting.
It seems to me that all issues are coming directly from ray/torch (and resulting in a structure full of `nans`), what versions of those packages you have installed?

```python
2022-11-09 15:50:29,427 ERROR trial_runner.py:958 -- Trial PPO_this-aviary-v0_924f1_00000: Error processing event.
Traceback (most recent call last):
File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/trial_runner.py", line 924, in _process_trial
results = self.trial_executor.fetch_result(trial)
File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/ray_trial_executor.py", line 787, in fetch_result
result = ray.get(trial_future[0], timeout=DEFAULT_GET_TIMEOUT)
File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/_private/client_mode_hook.py", line 105, in wrapper
return func(*args, **kwargs)
File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/worker.py", line 1713, in get
raise value.as_instanceof_cause()
ray.exceptions.RayTaskError(ValueError): ray::PPO.train() (pid=926055, ip=192.168.1.110, repr=PPO)
File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/agents/ppo/ppo_torch_policy.py", line 82, in loss
curr_action_dist = dist_class(logits, model)
File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/models/torch/torch_action_dist.py", line 185, in init
self.dist = torch.distributions.normal.Normal(mean, torch.exp(log_std))
File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/torch/distributions/normal.py", line 50, in init
super(Normal, self).init(batch_shape, validate_args=validate_args)
File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/torch/distributions/distribution.py", line 55, in init
raise ValueError(
ValueError: Expected parameter loc (Tensor of shape (128, 4)) of distribution Normal(loc: torch.Size([128, 4]), scale: torch.Size([128, 4])) to satisfy the constraint Real(), but found invalid values:
tensor([[nan, nan, nan, nan],
[nan, nan, nan, nan],
[nan, nan, nan, nan],
[nan, nan, nan, nan],
[nan, nan, nan, nan],
[nan, nan, nan, nan],
[nan, nan, nan, nan],
[nan, nan, nan, nan],
[nan, nan, nan, nan],
```

jackvice, 2022-11-11:
I've tested on one machine that has ray 1.9.0 and torch version 1.10.1 and on another with ray 1.9.0 and torch 1.11.0 and policy still diverging . :(     Thanks for any help.

JacopoPan, 2022-11-12:
I'd need to try and replicate the issue: can you list OS, CPU, (GPU/driver if appropriate), packages in your conda environment and a minimal example (e.g. by creating a PR) that would consistently lead to that error on your machine?

jackvice, 2022-11-12:
To produce the error, I set `"timesteps_total": 10000000,` at line 283 of multiagent.py and then run `python multiagent.py --act 'rpm'`.  The setup for the two machines I have tried are listed below.  Tried on GPU and CPU only on both machines.

Machine 1: Ubuntu 20.04.5 on Intel i7-10700F with GTX 1080 and RTX 2080 GPU's. Nvidia 460.106.00, CUDA 11.2
Conda packages:
`_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
async-timeout             4.0.2                    pypi_0    pypi
attrs                     22.1.0                   pypi_0    pypi
blas                      1.0                         mkl  
brotlipy                  0.7.0           py38h0a891b7_1004    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
ca-certificates           2022.9.24            ha878542_0    conda-forge
certifi                   2022.9.24          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py38h4a40e3a_1    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
click                     8.1.3                    pypi_0    pypi
cloudpickle               2.2.0                    pypi_0    pypi
contourpy                 1.0.5                    pypi_0    pypi
cryptography              37.0.1           py38h9ce1e76_0  
cudatoolkit               10.2.89             h713d32c_10    conda-forge
cycler                    0.10.0                   pypi_0    pypi
deprecated                1.2.13                   pypi_0    pypi
ffmpeg                    4.3                  hf484d3e_0    pytorch
filelock                  3.8.0                    pypi_0    pypi
fonttools                 4.38.0                   pypi_0    pypi
freetype                  2.12.1               hca18f0e_0    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.6.13               h85f3911_1    conda-forge
gputil                    1.4.0              pyh9f0ad1d_0    conda-forge
gym                       0.21.0                   pypi_0    pypi
gym-pybullet-drones       0.0.3                    pypi_0    pypi
idna                      3.4                pyhd8ed1ab_0    conda-forge
importlib-metadata        4.13.0                   pypi_0    pypi
intel-openmp              2021.4.0          h06a4308_3561  
jpeg                      9e                   h166bdaf_2    conda-forge
kiwisolver                1.4.4                    pypi_0    pypi
lame                      3.100             h166bdaf_1003    conda-forge
lcms2                     2.12                 hddcbb42_0    conda-forge
ld_impl_linux-64          2.39                 hc81fddc_0    conda-forge
lerc                      4.0.0                h27087fc_0    conda-forge
libdeflate                1.14                 h166bdaf_0    conda-forge
libffi                    3.4.2                h7f98852_5    conda-forge
libgcc-ng                 12.2.0              h65d4601_18    conda-forge
libgomp                   12.2.0              h65d4601_18    conda-forge
libiconv                  1.17                 h166bdaf_0    conda-forge
libnsl                    2.0.0                h7f98852_0    conda-forge
libpng                    1.6.38               h753d276_0    conda-forge
libsqlite                 3.39.4               h753d276_0    conda-forge
libstdcxx-ng              12.2.0              h46fd767_18    conda-forge
libtiff                   4.4.0                h55922b4_4    conda-forge
libuuid                   2.32.1            h7f98852_1000    conda-forge
libuv                     1.44.2               h166bdaf_0    conda-forge
libwebp-base              1.2.4                h166bdaf_0    conda-forge
libxcb                    1.13              h7f98852_1004    conda-forge
libzlib                   1.2.13               h166bdaf_4    conda-forge
lz4                       4.0.2                    pypi_0    pypi
matplotlib                3.6.1                    pypi_0    pypi
mkl                       2021.4.0           h06a4308_640  
mkl-service               2.4.0            py38h95df7f1_0    conda-forge
mkl_fft                   1.3.1            py38h8666266_1    conda-forge
mkl_random                1.2.2            py38h1abd341_0    conda-forge
ncurses                   6.3                  h27087fc_1    conda-forge
nettle                    3.6                  he412f7d_0    conda-forge
numpy                     1.23.3           py38h14f4228_0  
numpy-base                1.23.3           py38h31eccc5_0  
oauthlib                  3.2.2                    pypi_0    pypi
openh264                  2.1.1                h780b84a_0    conda-forge
openjpeg                  2.5.0                h7d73246_1    conda-forge
openssl                   3.0.5                h166bdaf_2    conda-forge
pillow                    9.2.0            py38ha3b2c9c_2    conda-forge
[... 140 lines cut ...]
scikit-image              0.19.3                   pypi_0    pypi
scipy                     1.9.1                    pypi_0    pypi
setuptools                65.4.1                   pypi_0    pypi
six                       1.16.0             pyhd3eb1b0_1
sqlite                    3.39.3               h5082296_0
stable-baselines3         1.6.1                    pypi_0    pypi
tabulate                  0.9.0                    pypi_0    pypi
[... cut ...]
typing_extensions         4.3.0            py38h06a4308_0
urllib3                   1.26.12          py38h06a4308_0
werkzeug                  2.2.2                    pypi_0    pypi
wheel                     0.37.1             pyhd3eb1b0_0
wrapt                     1.14.1                   pypi_0    pypi
xz                        5.2.6                h5eee18b_0
zipp                      3.8.1                    pypi_0    pypi
zlib                      1.2.12               h5eee18b_3
zstd                      1.5.2                ha4553b6_0`

JacopoPan, 2022-11-17:
I'm sorry, I haven't had the time to work on this: does the default example (as mentioned in the [paper](https://arxiv.org/pdf/2103.02142)) work as expected or that also results in `NaN`s now?

jackvice, 2022-11-20:
the default example from the paper does work with  "timesteps_total": 1200000.  I did a fresh pull today from the repo and it still breaks (NaN) when I pass --act 'rpm' after about an hour of training.  Only change to the code I made is the 
`"timesteps_total": 1200000`
on line 283 of multiagent.py

JacopoPan, 2022-11-25:
Can you take the learning agent out of the loop and running that same multi-agent environment (e.g. line 315 of `test_multiagent.py`, where the action from the policies is set) with a constant action (it should be a dictionary of size 4 arrays, with all zeros the drones should hover) to help me verify if it's a problem with the environment or the agent?

jackvice, 2022-11-26:
After training with `--act rpm`, I set line 315 to `action = {0: np.zeros(4), 1: np.zeros(4)}` and both drones appear to hover correctly.

JacopoPan, 2022-11-26:
I see, could you, at the same time, print out what is in `np.hstack([action[1], obs[1], obs[0]])` and returned by `..compute_single_action` in `temp[0]`, `temp[1]` to understand if there is something odd with the inputs or outputs of the networks (although I suspect the NaN's appear in one of the internal layers)?

jackvice, 2022-11-27:
Hi,  the NaN divergence happens during training and running test_multiagent on the save dir fails with:
`Traceback (most recent call last):
  File "test_multiagent.py", line 254, in <module>
    with open(ARGS.exp+'/checkpoint.txt', 'r+') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'results/save-leaderfollower-2-cc-kin-rpm 11.26.2022_16.29.04//checkpoint.txt'
`
The training NaN tensor 64 x 4 as show below:
`Failure # 1 (occurred at 2022-11-26_19-52-28)
Traceback (most recent call last):
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/trial_runner.py", line 924, in _process_trial
    results = self.trial_executor.fetch_result(trial)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/ray_trial_executor.py", line 787, in fetch_result
    result = ray.get(trial_future[0], timeout=DEFAULT_GET_TIMEOUT)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/_private/client_mode_hook.py", line 105, in wrapper
    return func(*args, **kwargs)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/worker.py", line 1713, in get
    raise value.as_instanceof_cause()
ray.exceptions.RayTaskError(ValueError): ray::PPO.train() (pid=828921, ip=192.168.1.110, repr=PPO)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/agents/ppo/ppo_torch_policy.py", line 82, in loss
    curr_action_dist = dist_class(logits, model)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/ray/rllib/models/torch/torch_action_dist.py", line 185, in __init__
    self.dist = torch.distributions.normal.Normal(mean, torch.exp(log_std))
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/torch/distributions/normal.py", line 50, in __init__
    super(Normal, self).__init__(batch_shape, validate_args=validate_args)
  File "/home/jack/anaconda3/envs/drones/lib/python3.8/site-packages/torch/distributions/distribution.py", line 55, in __init__
    raise ValueError(
ValueError: Expected parameter loc (Tensor of shape (64, 4)) of distribution Normal(loc: torch.Size([64, 4]), scale: torch.Size([64, 4])) to satisfy the constraint Real(), but found invalid values:
tensor([[nan, nan, nan, nan], 
tensor([[nan, nan, nan, nan], ...`

thx. All help is much appreciated.

JacopoPan, 2022-11-29:
This seems to come from `torch` when it finds a tensor with lists of 4 `NaN`'s, not the code in this repo itself. 
But I wonder if these `NaN`'s are wrongly logged `actions` coming from the `gym-pybullet-drones` environment.

Can you try to print out (or use a debugger) what `action` goes in and what `obs` is returned by method `.step()` (in `BaseAviary`) as training proceeds?

jackvice, 2022-12-15:
Running `python3 multiagent.py --env 'leaderfollower'`, the last set of actions and observation are below.   The policy diverges and spits out [nan] after 3 hours.  These are the last ten actions and observations from `.step()` in BaseAviary.
`{0: array([-0.21322441], dtype=float32), 1: array([-1.], dtype=float32)}, {0: array([ 0.        ,  0.        ,  0.10552978,  0.        , -0.        ,        0.        ,  0.        ,  0.        , -0.00090858,  0.        ,       0.        ,  0.        ]), 1: array([ 0.01058667,  0.01058667,  0.10189757,  0.        , -0.        , 0.        ,  0.        ,  0.        , -0.00389826,  0.        ,  0.        ,  0.        ])}
{0: array([0.39939213], dtype=float32), 1: array([1.], dtype=float32)}, {0: array([ 0.        ,  0.        ,  0.10553902,  0.        , -0.        ,
        0.        ,  0.        ,  0.        ,  0.00183648,  0.        ,
        0.        ,  0.        ]), 1: array([ 0.01058667,  0.01058667,  0.10190118,  0.        , -0.        ,
        0.        ,  0.        ,  0.        ,  0.00307836,  0.        , 0.        ,  0.        ])}
{0: array([-0.10729766], dtype=float32), 1: array([-1.], dtype=float32)}, {0: array([ 0.        ,  0.        ,  0.1055565 ,  0.        , -0.        ,        0.        ,  0.        ,  0.        ,  0.00110692,  0.        ,
        0.        ,  0.        ]), 1: array([ 0.01058667,  0.01058667,  0.10188988,  0.        , -0.        ,
        0.        ,  0.        ,  0.        , -0.00355743,  0.        ,  0.        ,  0.        ])}
{0: array([0.06007123], dtype=float32), 1: array([1.], dtype=float32)}, {0: array([ 0.        ,  0.        ,  0.1055734 ,  0.        , -0.        ,
        0.        ,  0.        ,  0.        ,  0.00151529,  0.        ,
        0.        ,  0.        ]), 1: array([ 0.01058667,  0.01058667,  0.10189774,  0.        , -0.        ,
        0.        ,  0.        ,  0.        ,  0.0034189 ,  0.        ,  0.        ,  0.        ])}
{0: array([-0.17268103], dtype=float32), 1: array([-1.], dtype=float32)}, {0: array([ 0.        ,  0.        ,  0.10558356,  0.        , -0.        ,       0.        ,  0.        ,  0.        ,  0.00034429,  0.        ,
        0.        ,  0.        ]), 1: array([ 0.01058667,  0.01058667,  0.1018907 ,  0.        , -0.        ,
        0.        ,  0.        ,  0.        , -0.00321716,  0.        , 0.        ,  0.        ])}
{0: array([-0.04746491], dtype=float32), 1: array([1.], dtype=float32)}, {0: array([ 0.00000000e+00,  0.00000000e+00,  1.05585441e-01,  0.00000000e+00,
       -0.00000000e+00,  0.00000000e+00,  0.00000000e+00,  0.00000000e+00,
        2.14683282e-05,  0.00000000e+00,  0.00000000e+00,  0.00000000e+00]), 1: array([ 0.01058667,  0.01058667,  0.10190281,  0.        , -0.        ,
        0.        ,  0.        ,  0.        ,  0.00375888,  0.        , 0.        ,  0.        ])}
{0: array([-0.063905], dtype=float32), 1: array([-1.], dtype=float32)}, {0: array([ 0.        ,  0.        ,  0.10558245,  0.        , -0.        ,
        0.        ,  0.        ,  0.        , -0.00041262,  0.        ,
        0.        ,  0.        ]), 1: array([ 0.01058667,  0.01058667,  0.10190002,  0.        , -0.        ,
        0.        ,  0.        ,  0.        , -0.00287747,  0.        ,  0.        ,  0.        ])}
{0: array([0.11206794], dtype=float32), 1: array([1.], dtype=float32)}, {0: array([ 0.        ,  0.        ,  0.10558303,  0.        , -0.        ,
        0.        ,  0.        ,  0.        ,  0.0003523 ,  0.        ,
        0.        ,  0.        ]), 1: array([ 0.01058667,  0.01058667,  0.10191638,  0.        , -0.        ,
        0.        ,  0.        ,  0.        ,  0.00409828,  0.        ,  0.        ,  0.        ])}
{0: array([-0.21531594], dtype=float32), 1: array([-1.], dtype=float32)}, {0: array([ 0.        ,  0.        ,  0.10557651,  0.        , -0.        ,       0.        ,  0.        ,  0.        , -0.00110497,  0.        ,
        0.        ,  0.        ]), 1: array([ 0.01058667,  0.01058667,  0.10191783,  0.        , -0.        ,
        0.        ,  0.        ,  0.        , -0.00253836,  0.        ,  0.        ,  0.        ])}
{0: array([-0.00944871], dtype=float32), 1: array([1.], dtype=float32)}, {0: array([ 0.        ,  0.        ,  0.10556222,  0.        , -0.        ,        0.        ,  0.        ,  0.        , -0.00116831,  0.        ,
        0.        ,  0.        ]), 1: array([ 0.01058667,  0.01058667,  0.10193842,  0.        , -0.        ,
        0.        ,  0.        ,  0.        ,  0.00443711,  0.        ,  0.        ,  0.        ])}`

zhaohubo, 2023-08-17:
hi, did you solve this problem?

jackvice, 2023-08-17:
No I didn't. is it still a problem in new versions?  I switched to Isaac gym.