learnsyslab/gym-pybullet-drones #128: Multiagent environment, error while terminating agents one by one

camillo2008, 2022-11-24:
Hi @JacopoPan,
I am struggling to understand how to terminate one agent but not the whole episode without getting an error in a multiagent environment.

In particular,  in the method **_computeDone** by default all agents are terminated after the seconds counter reach the maximum, something like this:
`        bool_val = True if self.step_counter/self.SIM_FREQ > self.EPISODE_LEN_SEC else False
        done = {i: bool_val for i in range(self.NUM_DRONES)}
        done["__all__"] = bool_val
        return done
`
Where for what i understand, set done=True to a single drone, means stop the single policy that guide the specific drone.
Where done["__all__"]=True mean that the episode is ended, and the reset method need to be called, this flag needs to be true when all the drones are done(set to True)

The problem is that: when I try to set the agent to done=True, one by one according to my problem, the code crash, this error is the encountered even in leader-follower and the other multiagent examples.

I have tried to figure it out on my own, but I have failed.

I have founded this useful issue: [https://github.com/ray-project/ray/issues/10761](url)
That talk about using one flag:"no_done_at_end": True, I am not sure that if I use this flag the policy stop tacking action when set to done.

Do you have any idea, for example, why in leader follower I encounter this problem?
I am trying to train multiple policy in the same problem, where one agent --> one policy, but some policy can die earlier than others.

Error encountered:
ValueError: ('Batches sent to postprocessing must only contain steps from a single trajectory.', SampleBatch(1002: ['obs', 'new_obs', 'actions', 'rewards', 'dones', 'infos', 'eps_id', 'unroll_id', 'agent_index', 'vf_preds', 'action_dist_inputs', 'action_logp']))

code to replicate in leader-follower while using multiagent:
`    def _computeDone(self):
        bool_val = True if self.step_counter / self.SIM_FREQ > self.EPISODE_LEN_SEC else False
        done = {i: bool_val for i in range(self.NUM_DRONES)}
        done[0] = True
        done["__all__"] = bool_val 
        return done
`

JacopoPan, 2022-11-25:
Hi @camillo2008 

the `done` flag is really just a boolean that informs an RL agent/library (like stable-baselines3, rllib) that the environment terminated and should be reset (`env.reset()`).

In the multiagent class from rllib, `done` is a dictionary of `num_agents`+1 booleans (with information about each agent and the whole environment).

If you explain to me what type of problem you are trying to formalize, I'm happy to try and suggest how to set those values but what you are dealing with it's either a formalization problem (to make your environment compatible with Ray/rllib) or an agent compatibility issue (and you might want to consider using a different RL library or implement your own).

camillo2008, 2022-11-28:
First of all, I'm sorry, but I'm having a hard time understanding RLib, also because RLib 1.9 documentation is missing.

So I'm trying to use as multi-agent base, to get something like challenge:[ICRA 2022 dodge drone challenge](https://uzh-rpg.github.io/icra2022-dodgedrone/).
Long story short, the problem is: many drones have to dodge static orbs in a defined space.
If the drone collides with the space wall or an obstacle, it should restart from a random position at the start.
So each drone can end its episode prematurely if it collides with something, I'm calculating the collision and when it happens I'm setting the specific drone to done=True in the _ComputeDone method. And then the pop-up error.

Now it's clear to me that the problem is a bad fit of the RL library to my problem, but I'm still trying to get it correct

JacopoPan, 2022-11-28:
I think it would help you understand the problem better if you solved the problem with a single agent first, maybe with a simpler/more documented library like stable-baselines3

closed 2022-11-29 by camillo2008