learnsyslab/gym-pybullet-drones #129: how can i study the drone model?

powehi2020, 2022-11-25:
hi~ I am a freshman to study control drone , where can i change the drone model in the code ?

JacopoPan, 2022-11-25:
Hi @powehi2020,

can you elaborate a bit? Do you want to change the controller or the drone's physical model?

powehi2020, 2022-11-28:
I want to change controller model . How can I do ?

powehi2020, 2022-11-28:
> Hi @powehi2020,
> 
> can you elaborate a bit? Do you want to change the controller or the drone's physical model?

or , the current simplepidcontrol and DSLPIDControl are from which controller model ?

JacopoPan, 2022-11-28:
The 2 PID controller do not leverage a prior dynamics model, just inertial information and other physical/motor-specific parameters (taken from system identification work that is references in the URDF files of each drone under `/assets`/).

The base implementation of each of the 2 PID controller is internal (for DSL) and from another open-source project (for simplePID), both are referenced in the source files of the the controllers.

If you are looking at other, model-based controllers (LQR, MPC), you should check out: https://github.com/utiasDSL/safe-control-gym

powehi2020, 2022-11-29:
> https://github.com/utiasDSL/safe-control-gym

Thank you for your answer !

gy2256, 2023-07-15:
I believe https://github.com/utiasDSL/safe-control-gym is limited 2d drone dynamics. It would be nice to provide the system dynamics so people can do model-based control.

JacopoPan, 2023-07-16:
Hi @gy2256 

it's true that `safe-control-gym` original paper is limited to 2D but there is a 3D version of it in the `main` branch now
https://github.com/utiasDSL/safe-control-gym/blob/140ed17dbd91d73a1f6537520b610adff732b9aa/safe_control_gym/envs/gym_pybullet_drones/quadrotor.py#L511