learnsyslab/gym-pybullet-drones #134: Where is the leader path defined in Multiagent.py using the leaderfollower aviary?

mattyboy-19, 2023-02-06:
I am just wondering how/where the path of leader is defined for the leader follower example within multiagent.py?

JacopoPan, 2023-02-07:
In the 1-D example in [this paper ](https://arxiv.org/pdf/2103.02142) the leader does not have a path but (because it is also a learning agent) it is rewarded for hovering at a given position in z
https://github.com/utiasDSL/gym-pybullet-drones/blob/7f930179ad9691229dbe4a4af53c4fea6e7db1fd/gym_pybullet_drones/envs/multi_agent_rl/LeaderFollowerAviary.py#L84

If you want the leader to be a non-learning agent, executing a scripted trajectory, you could put that in method
https://github.com/utiasDSL/gym-pybullet-drones/blob/7f930179ad9691229dbe4a4af53c4fea6e7db1fd/gym_pybullet_drones/envs/multi_agent_rl/BaseMultiagentAviary.py#L168

and reward the other agent for being close to the leader (but is it no longer a multi-agent RL problem..)

mattyboy-19, 2023-02-07:
Thanks for answering! 

So essentially, if I want a pre-defined path for a leader all I need to do is define it in the aviary? 
And even though it is single agent RL, I would still need to specify that there are two drones in the environment?

JacopoPan, 2023-02-07:
Yes,

you need to create an object of a class that extends `BaseAviary` with 2 drones, its `.step()` method will take a dictionary as argument with the 2 control inputs and return a dictionary as an obs, with states for both drones.

But for your problem, I'd wrap that object in such a way that half of each action/obs dictionary you use to control the scripted drone and half you plug to a single agent RL algorithm. You will have to define a reward function computed from the states of the 2 drones, I assume.

mattyboy-19, 2023-02-08:
Ok, thank you for your help!

closed 2023-02-08 by mattyboy-19