learnsyslab/gym-pybullet-drones #139: Feature Request: migrate from gym to gymnasium

elliottower, 2023-03-17:
Hi, would it be possible for pybullet-drones to be upgraded from gym to gymnasium? Gymnasium is the maintained version of openai gym and is compatible with current RL training libraries ([rllib](https://github.com/ray-project/ray/tree/master/rllib) and [tianshou](https://github.com/thu-ml/tianshou) have already migrated, and [stable-baselines3](https://github.com/DLR-RM/stable-baselines3/blob/e5deeed16efb57c34ccdcb14692439154d970527/docs/guide/install.rst#bleeding-edge-version) will soon).

This repository is currently listed in the gymnasium [third party environments](https://gymnasium.farama.org/environments/third_party_environments/) but we are cleaning the list up to only include maintained gymnasium-compatible repositories.

For information about upgrading and compatibility, see [migration guide](https://gymnasium.farama.org/content/migration-guide/) and [gym compatibility](https://gymnasium.farama.org/content/gym_compatibility/).

I see there's been a little discussion in the issues and PRs about migration (https://github.com/utiasDSL/gym-pybullet-drones/pull/136, https://github.com/utiasDSL/gym-pybullet-drones/issues/133), just wanted to check if there were any plans to migrate.

JacopoPan, 2023-03-20:
Yes, PR #138 is now the one checking if the interface migration works with the more recent `rllib` scripts.

JacopoPan, 2023-12-08:
This is now completed

closed 2023-12-11 by elliottower