learnsyslab/gym-pybullet-drones #141: Multiagent drone gives continues error and not able to execute and see the how it works

sanoufar, 2023-03-22:
This is error that i gets when try to excute the multiagent drone  

Desktop/gym-pybullet-drones-master/experiments/learning$ python multiagent.py --num_drones 2 --env flock --obs rgb --act one_d_pid --algo cc --num_workers 0
pybullet build time: May 20 2022 19:44:17
usage: multiagent.py [-h] [--num_drones] [--env] [--obs] [--act] [--algo]
                     [--workers]
multiagent.py: error: unrecognized arguments: --num_workers 0

JacopoPan, 2023-03-28:
Hi @sanoufar 

my bad, the argument is actually called `--workers`, not `--num_workers`

https://github.com/utiasDSL/gym-pybullet-drones/blob/c997c6bc73e9a623bdfe6fcaf9c4d1303cf14dfc/experiments/learning/multiagent.py#L149

JacopoPan, 2023-03-28:
Should be fixed now with [6b88951](https://github.com/utiasDSL/gym-pybullet-drones/commit/6b88951c8a23ae1dba037c3ef53d3c3389e3e4af)