learnsyslab/gym-pybullet-drones #144: multiagent.py and test_multiagent don't support more than 2 drones.

jackvice, 2023-04-04:
It would appear that multiagent.py and test_multiagent.py are partially hard coded for only 2 drones.  Is this intentional or am I missing something.  for example:
`
  #### Set up the multiagent params of the trainer's config ##
    config["multiagent"] = { 
        "policies": {
            "pol0": (None, observer_space, action_space, {"agent_id": 0,}),
            "pol1": (None, observer_space, action_space, {"agent_id": 1,}),
        },
        "policy_mapping_fn": lambda x: "pol0" if x == 0 else "pol1", # # Function mapping agent ids to policy ids
        "observation_fn": central_critic_observer, # See rllib/evaluation/observation_function.py for more info
    }
`

BCWang93, 2023-04-05:
> 

hi, did you solve this problem?

jackvice, 2023-04-05:
Almost, I kinda have 3 drones working in leader follower but I am seeing divergent behavior.  I am confused by this code in the main loop of test_multiagent.  I added a third drone and the original 2 drone code is commented out:
` 
for i in range(16*int(test_env.SIM_FREQ/test_env.AGGR_PHY_STEPS)): # Up to 6''
        #### Deploy the policies ###################################
        temp = {}
        #temp[0] = policy0.compute_single_action(np.hstack([action[1], obs[1], obs[0]])) # Counterintuitive order, check params.json
        #temp[1] = policy1.compute_single_action(np.hstack([action[0], obs[0], obs[1]]))
        
        temp[0] = policy0.compute_single_action(np.hstack([action[2], obs[2], obs[1]])) # Counterintuitive order, check params.json
        temp[1] = policy1.compute_single_action(np.hstack([action[1], obs[1], obs[0]]))
        temp[2] = policy2.compute_single_action(np.hstack([action[0], obs[0], obs[1]])) # 3rd drone addition
        action = {0: temp[0][0], 1: temp[1][0], 2: temp[2][0]}
`

JacopoPan, 2023-04-12:
The 2-drone system is by design (as the scirpt was based on a "centraliezed critic/decentralized actor" example for Rllib.

To use it with more drones you do not only need to duplicate the policies (as @jackvice is hinting at, I think) but also re-write the centralized and related classes/methods:
https://github.com/utiasDSL/gym-pybullet-drones/blob/6b88951c8a23ae1dba037c3ef53d3c3389e3e4af/experiments/learning/multiagent.py#L69
https://github.com/utiasDSL/gym-pybullet-drones/blob/6b88951c8a23ae1dba037c3ef53d3c3389e3e4af/experiments/learning/multiagent.py#L110
https://github.com/utiasDSL/gym-pybullet-drones/blob/6b88951c8a23ae1dba037c3ef53d3c3389e3e4af/experiments/learning/multiagent.py#L124

Although I wrote that example to showcase the multi-agent compatibility with Rllib, these days I would probably recommend a fully centralized (1 environment with N drones = 1 RL agent) or decentralized (1 environment with N drones = N independent RL agents of N copies of the same agent) approach.