learnsyslab/gym-pybullet-drones #147: hover at the target location

zehuiWu0306, 2023-05-10:


zehuiWu0306, 2023-05-10:
def _computeObs(self):
        """the method to compute the current observation vector based on the normalized Obs."""
        normalized_obs = super()._computeObs()
        max_xy = self.MAX_XY
        max_z = self.MAX_Z
        normalized_target_pos = np.array([(self.target_position[0] / max_xy),
                                          (self.target_position[1] / max_xy),
                                          (self.target_position[2] / max_z)
                                          ])
        pos_diff = [normalized_target_pos[:3] - normalized_obs[:3]]
        velocity = normalized_obs[3:6]
        if np.linalg.norm(velocity) == 0:
            velocity_direction = velocity / 1
        else:
            velocity_direction = velocity / np.linalg.norm(velocity)
        target_direction = pos_diff / np.linalg.norm(pos_diff)
        angle = np.arccos(np.clip(np.dot(target_direction, velocity_direction), -1.0, 1.0))
        self.angle = angle 
obs = np.hstack([pos_diff[0], normalized_obs[3:]])
        self.obs = obs
        return obs
 def _computeReward(self):
        """Computes the current reward value.

        Returns
        -------
        float
            The reward.

        """
        current_distance = np.linalg.norm(self.obs[:3])

        if self.prev_distance is None:
            self.prev_distance = current_distance

        distance_difference = self.prev_distance - current_distance

        self.prev_distance = current_distance

        big_reward = 5.0
        distance_threshold = 0.001

        if current_distance < distance_threshold:
            reward = big_reward
            self.waypoint_is_reached = True
        else:
            velocity_reward = 0.1 * np.cos(self.angle)
            reward = 2 * distance_difference + velocity_reward
        return reward

    ################################################################################

    def _computeDone(self):
        """Computes the current done value.

        Returns
        -------
        bool
            Whether the current episode is done.

        """
        if self.waypoint_is_reached or self.step_counter > 50000:
            return True
        else:
            return False

zehuiWu0306, 2023-05-10:
The reward value does not converge and the test results show that the intelligent body does not hover as envisaged

zehuiWu0306, 2023-05-10:
If anyone can answer any questions, I would appreciate it!

JacopoPan, 2023-05-14:
Hi @zehuiWu0306 

if you posted your entire codebase (as a fork and/or PR) and steps to re-produce your problem, I could try to look into it, judging from just static analysis of code snippets is non-trivial.