learnsyslab/gym-pybullet-drones #153: Z-axis trajectory not changing after editing fly.py example

dhypoon, 2023-06-12:
I was trying to input custom trajectories into the fly.py example and realized that even after changing the z-axis TARGET_POS in the code, it wouldn't do anything in the simulation. I notice that I can edit the initial z-position of the drones and I can't seem to find any ceiling limiting the z-position, so I don't know why this isn't working. Does anyone have any ideas?

JacopoPan, 2023-06-18:
Hi @dhypoon 

in the `fly.py` example the target/reference z passed to the controllers is the initial one.

Change this line to modify the behavior.
https://github.com/utiasDSL/gym-pybullet-drones/blob/6b88951c8a23ae1dba037c3ef53d3c3389e3e4af/gym_pybullet_drones/examples/fly.py#L171

abdul-mannan-khan, 2023-07-10:
Hello @dhypoon,

Could you fix your problem. I am still struggling to do it. Could you please share what trajectory you tested and where you defined it and where you passed it so that it could be tracked? Would just uncommenting line 171 be enough? Thank you.

JacopoPan, 2023-07-16:
@abdul-mannan-khan 

it's not about uncommenting 171, it's `INIT_XYZS[j, 2]` that needs to contain the desired z position