learnsyslab/gym-pybullet-drones #155: Why pybullet cannot accurately express the dimensions directly defined by urdf

Hobbit-wgh, 2023-06-16:
When we directly specify a cylinder with a length of 500 and a radius of 10 in URDF, it is often larger than this size in pybullet.

such as：

  <link name="FT_link">
    <inertial>
      <mass value="0.15"/>
      <origin rpy="0 0 0" xyz="0.0 0.0 -0"/>
      <inertia ixx="0.0" ixy="0.0" ixz="0.0" iyy="0" iyz="0.0" izz="0"/>
    </inertial>
    <visual>
      <geometry>
        <cylinder length="0.05" radius="0.01"/>
      </geometry>
      <material name="white">
        <color rgba="0.7 0.7 0.7 1.0"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <cylinder length="0.05" radius="0.01"/>
      </geometry>
    </collision>
  </link>

but，in  pybullet this cylinder may be 550 and 105

JacopoPan, 2023-06-18:
Hi @Hobbit-wgh 

I haven't observed this behavior (do you have a reproducible example) but, if you think it is caused by Bullet, it would be resulting from code outside of this repo.

One note I would have is that the `visual` and `collision` geometries are completely separate ones.

Hobbit-wgh, 2023-06-19:
> Hi @Hobbit-wgh
> 
> I haven't observed this behavior (do you have a reproducible example) but, if you think it is caused by Bullet, it would be resulting from code outside of this repo.
> 
> One note I would have is that the `visual` and `collision` geometries are completely separate ones.
I want to insert the 20mm (urdf directly defined) axis into the 21mm hole (stl file import), but the axis can not be inserted

JacopoPan, 2023-06-19:
Hi @Hobbit-wgh 

are you sure you are on the right repo? I'm not sure what 20mm and 21mm elements you are talking about.

closed 2023-06-19 by JacopoPan