learnsyslab/gym-pybullet-drones #158: unable to run multiagent.py

abdul-mannan-khan, 2023-07-03:
Hello Jacob, 

Thank you for your repository with wonderful discuss at ICRA London. I am having a very basic problem if you could put some light on it. I am trying to run `multiagent.py` but getting following error. 

`/home/yf23868/.local/lib/python3.8/site-packages/gym/spaces/box.py:73: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(
Traceback (most recent call last):
  File "python/ray/_raylet.pyx", line 380, in ray._raylet.prepare_args
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/ray/serialization.py", line 361, in serialize
    return self._serialize_to_msgpack(value)
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/ray/serialization.py", line 336, in _serialize_to_msgpack
    msgpack_data = MessagePackSerializer.dumps(value, _python_serializer)
  File "python/ray/includes/serialization.pxi", line 162, in ray._raylet.MessagePackSerializer.dumps
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/msgpack/__init__.py", line 47, in packb
    return Packer(**kwargs).pack(o)
TypeError: __init__() got an unexpected keyword argument 'strict_types'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "multiagent.py", line 289, in <module>
    results = tune.run(
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/tune.py", line 597, in run
    runner.step()
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/trial_runner.py", line 708, in step
    self.trial_executor.stage_and_update_status(self._live_trials)
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/ray_trial_executor.py", line 264, in stage_and_update_status
    if not self._pg_manager.stage_trial_pg(trial):
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/utils/placement_groups.py", line 410, in stage_trial_pg
    return self._stage_pgf_pg(pgf)
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/ray/tune/utils/placement_groups.py", line 424, in _stage_pgf_pg
    self._staging_futures[pg.ready()] = (pgf, pg)
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/ray/util/placement_group.py", line 74, in ready
    return bundle_reservation_check.options(
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/ray/remote_function.py", line 181, in remote
    return func_cls._remote(
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/ray/util/tracing/tracing_helper.py", line 295, in _invocation_remote_span
    return method(self, args, kwargs, *_args, **_kwargs)
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/ray/remote_function.py", line 346, in _remote
    return invocation(args, kwargs)
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.8/site-packages/ray/remote_function.py", line 328, in invocation
    object_refs = worker.core_worker.submit_task(
  File "python/ray/_raylet.pyx", line 1444, in ray._raylet.CoreWorker.submit_task
  File "python/ray/_raylet.pyx", line 1448, in ray._raylet.CoreWorker.submit_task
  File "python/ray/_raylet.pyx", line 388, in ray._raylet.prepare_args
TypeError: Could not serialize the argument <ray.util.placement_group.PlacementGroup object at 0x7f3368c04be0> for a task or actor ray.util.placement_group._export_bundle_reservation_check_method_if_needed.<locals>.bundle_reservation_check_func. Check https://docs.ray.io/en/master/serialization.html#troubleshooting for more information.`

Could you give me some directions to solve it? Thank you.

JacopoPan, 2023-07-16:
Hi @abdul-mannan-khan 

the problem you are facing seems to come from Ray/rllib and I've been moving away the repo from that dependency.
I'm currently working on this branch https://github.com/utiasDSL/gym-pybullet-drones/tree/main that uses `gymnasium` (as a replacement for `gym`) and `stable-baselines3` 2.0 (instead of the older `stable-baselines3` 1.x and `rllib`)

abdul-mannan-khan, 2023-07-17:
Hello JacoboPan, 

Thank you so much for your response. I cannot see that folder/file in your repository. Is it public? Thank you.

JacopoPan, 2023-07-17:
the `main` branch does not have a `multiagent.py` script because the action/observations spaces are now ndarrays (instead of dictionaries) so even multi agent aviaries can used with stable-baselines3 (as in learn.py) instead of rllib

abdul-mannan-khan, 2023-07-17:
Hello @JacopoPan, 

Thank you for your response. I tried changing 

https://github.com/utiasDSL/gym-pybullet-drones/blob/183c84a56c3f8a941fd7ec6c3e4308532b65b05e/pyproject.toml#L19C1-L19C14

to gymnasium 0.28 and tried replacing `import gymnasium as gym`. I am getting many errors. In deed, in the whole code section wherever, gym is being used, I am replacing it with gymnasium. But there are many sub-classes/functions which seems have changed name in gymnasium. Would you mind sharing the link of repository in which you are using gymnasium. Thank you so much.

JacopoPan, 2023-07-17:
It's branch main of this same repo https://github.com/utiasDSL/gym-pybullet-drones/tree/main

abdul-mannan-khan, 2023-07-17:
Hello @JacopoPan 

Thank you for your quick response. I cannot see `/gum-pybullet-drones/tree/main`. Please correct me if I am doing something wrong.

JacopoPan, 2023-07-17:
it's not a folder, it's another branch, just clone with

```sh
git clone -b main https://github.com/utiasDSL/gym-pybullet-drones.git
```

abdul-mannan-khan, 2023-07-17:
It worked now. Thank you @JacopoPan for your time.

closed 2023-07-17 by abdul-mannan-khan