learnsyslab/gym-pybullet-drones #172: mutliagent.py allow 3D flight?

zhaohubo, 2023-09-22:
#126 As mentioned in this, does it support 3D flight？

zhaohubo, 2023-10-10:
Hi @JacopoPan 
I have successfully conducted experiments with two drones. Now, I want to expand my experiments by including more drones, for example, three in total. Here is the modified code I have implemented. When the ActionType is set to "one_d_rpm", the leader-follower functionality works successfully. However, when I try to implement 3D flight by changing the ActionType to "rpm/pid/vel", the training results are not satisfactory. I'm wondering if there is an error in my model modifications or if the algorithm parameters need to be adjusted. It would be greatly appreciated if you could assist me with this as it is crucial for my project.
`
    import sys
    import os
    import time
    import argparse
    from datetime import datetime
    import subprocess
    import pdb
    import math
    import numpy as np
    import pybullet as p
    import pickle
    import matplotlib.pyplot as plt
    import gym
    from gym import error, spaces, utils
    from gym.utils import seeding
    from gym.spaces import Box, Dict
    import torch
    import torch.nn as nn
    from ray.rllib.models.torch.fcnet import FullyConnectedNetwork
    import ray
    from ray import tune
    from ray.tune.logger import DEFAULT_LOGGERS
    from ray.tune import register_env
    from ray.rllib.agents import ppo
    from ray.rllib.agents.ppo import PPOTrainer, PPOTFPolicy
    from ray.rllib.examples.policy.random_policy import RandomPolicy
    from ray.rllib.utils.test_utils import check_learning_achieved
    from ray.rllib.agents.callbacks import DefaultCallbacks
    from ray.rllib.models.torch.torch_modelv2 import TorchModelV2
    from ray.rllib.models import ModelCatalog
    from ray.rllib.policy.sample_batch import SampleBatch
    from ray.rllib.env.multi_agent_env import ENV_STATE

    sys.path.append(os.path.dirname(os.path.dirname(os.path.dirname(os.path.abspath(__file__)))))
    from gym_pybullet_drones.envs.BaseAviary import DroneModel, Physics
    from gym_pybullet_drones.envs.multi_agent_rl.FlockAviary import FlockAviary
    from gym_pybullet_drones.envs.multi_agent_rl.LeaderFollowerAviary import LeaderFollowerAviary
    from gym_pybullet_drones.envs.multi_agent_rl.MeetupAviary import MeetupAviary
    from gym_pybullet_drones.envs.single_agent_rl.BaseSingleAgentAviary import ActionType, ObservationType
    from gym_pybullet_drones.utils.Logger import Logger

    import shared_constants

    OWN_OBS_VEC_SIZE = None  # Modified at runtime
    ACTION_VEC_SIZE = None  # Modified at runtime

    class CustomTorchCentralizedCriticModel(TorchModelV2, nn.Module):

        def __init__(self, obs_space, action_space, num_outputs, model_config, name):
            TorchModelV2.__init__(self, obs_space, action_space, num_outputs, model_config, name)
            nn.Module.__init__(self)
            self.action_model = FullyConnectedNetwork(
                Box(low=-1, high=1, shape=(OWN_OBS_VEC_SIZE,)),
                action_space,
                num_outputs,
                model_config,
                name + "_action"
            )
            self.value_model = FullyConnectedNetwork(
                obs_space,
                action_space,
                1,
                model_config,
                name + "_vf"
            )
            self._model_in = None

[... 224 lines cut ...]
    }

    #### Train #################################################
    results = tune.run(
        "PPO",
        stop=stop,
        config=config,
        verbose=True,
        checkpoint_at_end=True,
        local_dir=filename,
    )
    # check_learning_achieved(results, 1.0)

    #### Save agent ############################################
    checkpoints = results.get_trial_checkpoints_paths(trial=results.get_best_trial('episode_reward_mean',
                                                                                   mode='max'
                                                                                   ),
                                                      metric='episode_reward_mean'
                                                      )
    with open(filename + '/checkpoint.txt', 'w+') as f:
        f.write(checkpoints[0][0])

    #### Shut down Ray #########################################
    ray.shutdown()
`

JacopoPan, 2023-10-12:
Hi @zhaohubo 

I haven't tried a 3 drone example but, as you are doing, of course, you should start by extending the centralized critic.
If it works in the 1D case, I'd think that that part is correct, however, the 3D case with `rpm` inputs might be too difficult, what are the results with `pid`? Have you modified the reward function as well?