learnsyslab/gym-pybullet-drones #173: pybullet fails to install on python 3.10 venv

amithm1111, 2023-09-26:
pybullet fails to install on a python 3.10 venv leading to the failure of installation of gym-pybullet-drones. Related issue : https://stackoverflow.com/questions/76616039/how-to-resolve-errors-installing-pybullet

JacopoPan, 2023-10-09:
Hello @amithm1111 

I just installed `gym-pybullet-drones` (and the corresponding dependencies, including `pybullet`) in a Python 3.10 condo environment. Is this issue still current? If yes, can you provide you terminal output?

amithm1111, 2023-10-09:
It works with conda virtual environment. My issue is that it does not work in python 3.10 environment created using 'venv'. 

The error I'm getting :
```
ERROR: Failed building wheel for pybullet
    Running setup.py clean for pybullet
Failed to build pybullet
ERROR: could not build wheels for pybullet
```

I was able to find this workaround:
"The current version pybullet 3.2.5 [provides wheels](https://pypi.org/project/pybullet/3.2.5/#files) for Linux on x64 for Pythons up to Python 3.9. Use Python 3.9".

I was able to install gym-pybullet-drones in a python 3.9 venv (after editing pyproject.toml to `python = "^3.9"`

closed 2023-10-09 by amithm1111