learnsyslab/gym-pybullet-drones #178: unable to see logger file output at output_folder for learn.py

abdul-mannan-khan, 2023-11-07:
Hello, 
I am currently working on learn.py, and I have been facing an issue with viewing the learning results. Even though I have enabled output results, I am only getting a single picture with multiple graphs, which is not providing the detailed learning information I need.

I would greatly appreciate your guidance on how to access more detailed learning results. Is there a specific setting or configuration I should be aware of to obtain the desired level of detail? Here is what I am doing. Also, no recorded video. 

`python3 learn.py --guid 'True' --record_video 'True' --output_folder 'results'`

Thank you for your assistance.

JacopoPan, 2023-11-19:
Which branch are you working on @abdul-mannan-khan ?

The `learn.py` script is meant as a toy example of a complete RL training and replay workflow but for longer, more customized (with checkpoints) training runs, it would be better to use 
https://github.com/utiasDSL/gym-pybullet-drones/blob/paper/experiments/learning/singleagent.py (for training)
and https://github.com/utiasDSL/gym-pybullet-drones/blob/paper/experiments/learning/test_singleagent.py (for testing)

abdul-mannan-khan, 2023-11-20:
Thank you @JacopoPan for your response. The problem with [https://github.com/utiasDSL/gym-pybullet-drones/blob/paper/experiments/learning/singleagent.py](https://github.com/utiasDSL/gym-pybullet-drones/blob/paper/experiments/learning/singleagent.py) is that it is using `gym` which is causing problems. I tried modifying this code but it became very messey. Can you take a look please? Thank you so much.

JacopoPan, 2023-11-20:
Yes, I am working on a version of `learn.py` in branch `dev-sb3` that does most of the things that `singleagent.py` was for but using `gymnasium` instead of the old `gym`