learnsyslab/gym-pybullet-drones #179: ModuleNotFoundError: No module named 'gym_pybullet_drones'

ybl998877, 2023-11-13:
when I run 'pid.py' ,report 'ModuleNotFoundError: No module named 'gym_pybullet_drones''.The model name and path are right,i dont konw why this error occurs.

abdul-mannan-khan, 2023-11-14:
You need to 

`cd gym-pybullet-drones/`

```
conda create -n drones python=3.10
conda activate drones

pip3 install --upgrade pip
pip3 install -e . # if needed, `sudo apt install build-essential` to install `gcc` and build `pybullet`
```
If you will not run

`pip3 install -e . # if needed, `sudo apt install build-essential` to install `gcc` and build `pybullet``
from inside the directory 

`cd gym-pybullet-drones/`
 you will get the above mentioned error

ybl998877, 2023-11-17:
Thanks very much! I run it successfully.

JacopoPan, 2023-11-19:
Thanks @abdul-mannan-khan for answering before I could :D

closed 2023-11-19 by JacopoPan