learnsyslab/gym-pybullet-drones #181: PX4 integration

petergerten, 2023-12-02:
Did anybody integrate PX4 SITL with this framework to utilize it for higher level objectives?

dmvinson, 2024-07-12:
What would be involved in integrating with PX4? Is that covered under the `Use SITL services with steppable simulation` roadmap item? I'd be interested in this for my work, happy to contribute if there is an initial set of pointers to being development

JacopoPan, 2024-07-14:
Hi @dmvinson 

if you want to try and implement a lockstep version of PX4 SITL paired with one of the environments in this repo, it would be great and I'm happy to advise/guide/PR review the process.

I would start from the example in [`beta.py`](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/examples/beta.py) for which the execution instructions are [here](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/README.md#betaflight-sitl-example-ubuntu-only).

In this case the SITL is **NOT** in lockstep (the PyBullet simulation and the SITL of Betaflight run in parallel in separate processes
https://github.com/utiasDSL/gym-pybullet-drones/blob/3d7b12edd4915a27e6cec9f2c0eb4b5479f7735e/gym_pybullet_drones/envs/BetaAviary.py#L85-L89
and communication happens over UDP
https://github.com/utiasDSL/gym-pybullet-drones/blob/3d7b12edd4915a27e6cec9f2c0eb4b5479f7735e/gym_pybullet_drones/envs/BetaAviary.py#L158-L163

I believe that something similar using the default UDP/TCP port of SITL PX4 (see [this diagram](https://docs.px4.io/main/en/simulation/#sitl-simulation-environment)) and **WITH THE ADDITION** of lockstep execution (see [this paragraph](https://docs.px4.io/main/en/simulation/#lockstep-simulation) could be implemented by creating a new example `gym_pybullet_drones/examples/px4.py` and a new environment (spawning and talking to PX4 SITL) `gym_pybullet_drones/envs/PX4Aviary.py`

dmvinson, 2024-07-14:
Thanks for the guidance, I'll take a look at this over the next few days. I appreciate your help!