learnsyslab/gym-pybullet-drones #182: Cannot run simulation on linux Fedora

LucasPeters00, 2023-12-08:
When i try to run an example script it fails to run the simulation. I use Linux Fedora on my macbook. The code is tested for ubuntu but I can't dual boot my laptop with ubuntu. Is there a way to work around this problem or is it not possible with Fedora? I tried installing the missing package manually but that does not do anything.

the error:

(drones) lucaspeters@lucaspeters-macbookpro:~/Desktop/PDM/gym-pybullet-drones/gym_pybullet_drones/examples$ python3 pid.py
pybullet build time: Dec  7 2023 21:58:45
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
MESA-LOADER: failed to open asahi: /home/lucaspeters/.conda/envs/drones/lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib64/dri/asahi_dri.so) (search paths /usr/lib64/dri, suffix _dri)
failed to load driver: asahi
MESA-LOADER: failed to open swrast: /home/lucaspeters/.conda/envs/drones/lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib64/dri/swrast_dri.so) (search paths /usr/lib64/dri, suffix _dri)
failed to load driver: swrast
Creating context
Failed to create GL 3.3 context ... using old-style GLX context
Failed to create an OpenGL context

JacopoPan, 2023-12-08:
Hi @LucasPeters00 

this is really a video driver/OpenGL issue (used by PyBullet, which is turn used by this repo), one thing you can try, if you are using an NVIDIA card is: if you receive a "Failed to create and OpenGL context" message, launch nvidia-settings and under "PRIME Profiles" select "NVIDIA (Performance Mode)", reboot and try again.

LucasPeters00, 2023-12-08:
Hi @JacopoPan, Thanks for your reply. I'm not using a NVIDIA card but a multi-core GPU, I dual booted my macbook M1. So i guess there is no fix then for the driver/OpenGL issue.

LucasPeters00, 2023-12-11:
I have now downloaded into Ubuntu, and still got the same error:

(drones) ubuntu@localhost:~/Desktop/gym-pybullet-drones/gym_pybullet_drones/examples$ python3 pid.py
pybullet build time: Dec 11 2023 16:27:58
[INFO] BaseAviary.__init__() loaded parameters from the drone's .urdf:
[INFO] m 0.027000, L 0.039700,
[INFO] ixx 0.000014, iyy 0.000014, izz 0.000022,
[INFO] kf 0.000000, km 0.000000,
[INFO] t2w 2.250000, max_speed_kmh 30.000000,
[INFO] gnd_eff_coeff 11.368590, prop_radius 0.023135,
[INFO] drag_xy_coeff 0.000001, drag_z_coeff 0.000001,
[INFO] dw_coeff_1 2267.180000, dw_coeff_2 0.160000, dw_coeff_3 -0.110000
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=Mesa
GL_RENDERER=Apple M1 (G13G B1)
GL_VERSION=3.3 (Core Profile) Mesa 24.0~pre20231121-0asahi1.23.10
GL_SHADING_LANGUAGE_VERSION=3.30
pthread_getconcurrency()=0
Version = 3.3 (Core Profile) Mesa 24.0~pre20231121-0asahi1.23.10
Vendor = Mesa
Renderer = Apple M1 (G13G B1)
GL error 0x500 detected in glGenBuffers
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
viewMatrix (-0.8660253882408142, -0.2499999850988388, 0.4330126941204071, 0.0, 0.0, 0.8660253882408142, 0.5, 0.0, -0.4999999701976776, 0.4330126941204071, -0.75, 0.0, -5.21540641784668e-08, -7.450580596923828e-09, -3.0, 1.0)
projectionMatrix (1.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, -1.0000200271606445, -1.0, 0.0, 0.0, -0.02000020071864128, 0.0)
/home/ubuntu/anaconda3/envs/drones/lib/python3.10/site-packages/gymnasium/spaces/box.py:130: UserWarning: WARN: Box bound precision lowered by casting to float32
  gym.logger.warn(f"Box bound precision lowered by casting to {self.dtype}")
ven = Mesa
ven = Mesa

Are there options to adjust the graphics and load the rendering with something else? I tried some suggestions from other issues but nothing worked yet.

JacopoPan, 2023-12-12:
@LucasPeters00 this really looks like an issue with your OpenGL installation but I can assure you that PyBullet (and it's OpenGL renderer GUI) work well on both Ubuntu and Apple Silicon/macOS (because that is what I am using as well) but I never tried with Ubuntu on Apple Silicon.

LucasPeters00, 2023-12-12:
@JacopoPan What kind of graphics driver do you use for the simulation? For me creating the openGL context and loading the swrast driver fails maybe it has to do with a different version

JacopoPan, 2023-12-12:
@LucasPeters00 

On macOS, PyBullet simply uses
```
Version = 4.1 Metal - 88
Vendor = Apple
Renderer = Apple M2 Max
```

On Ubuntu (for me, on Lenovo with Intel CPU, Nvidia GPU), it creates GL 3.3 context
```
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=NVIDIA RTX ...
GL_VERSION=3.3.0 NVIDIA 535.129.03
...
```

It really seems that you are having problems running PyBullet https://github.com/bulletphysics/bullet3 (the physics engine and GUI used in this package), not this package per se

natema, 2024-02-12:
I was getting the same error of the OP on Fedora 39. 
Conda was trying to use `libstdc++.so.6.0.29` (in `~/.conda/envs/drones/lib`). 
My system a the more recent `libstdc++.so.6.0.32` in `/usr/lib`. 
Linking to my system's version solved the issue: 
```
ln -fs /usr/lib64/libstdc++.so.6 ~/.conda/envs/drones/lib/libstdc++.so.6
```

marques576, 2024-05-17:
> conda install -c conda-forge libstdcxx-ng=12

This worked for me in arch linux