learnsyslab/gym-pybullet-drones #184: Question about action buffer

phuongboi, 2023-12-30:
Hi, Thank you for your great work.
I am doing some experiments with reinforcement learning control, I don't know how the [action buffer](https://github.com/utiasDSL/gym-pybullet-drones/blob/2c2470c2043eb4cf474d9af2681c9fa10f2d0e2f/gym_pybullet_drones/envs/BaseRLAviary.py#L317) work. I tried observation space with 12 input (no action buffer) and it still work.

JacopoPan, 2024-01-11:
It is simply a concatenation to the state of the actions taken by the drones in the last half second. Depending on the type of application and the size of the networks being used, the intent is to potentially speed-up/stabilize training (but it is not strictly necessary).

phuongboi, 2024-01-12:
Thank your for your explanation