learnsyslab/gym-pybullet-drones #185: Custom training environment

wagh311, 2024-01-05:
Hello! I want to use DRL to train multiple drones to track moving targets in obstacle environments, so I need to customize the training environment. How can I implement a custom environment? For example, specify the size of the environment or add obstacles at specific locations.

JacopoPan, 2024-01-11:
Hi @wagh311 

just make a copy of class [`MultiHoverAviary`](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/envs/MultiHoverAviary.py), add the proper import/registrations in the init files ([link](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/envs/__init__.py), [link](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/__init__.py))
Obstacles are added in the `_addObstacles()` method of [BaseRLAviary.py](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/envs/BaseRLAviary.py) but that is only called for RGB state spaces, you can simply do something similar in the init (and/or override the `reset`) in your new class.

wagh311, 2024-01-11:
Hi @JacopoPan 
Thank you for your suggestion. I would also like to ask, does gym-pybullet-drones support the domain randomization function?

JacopoPan, 2024-01-11:
It should be straightforward to randomize the position of obstacles and the initial poses of the drones (from the init of the environment class), however, to randomize the physical properties (mass, inertia matrix) of the drones, you should consider that they are read from the URDF files in the `assets/` folder (you could overwrite them in the code of `BaseAviary`)

wagh311, 2024-01-11:
OK, I will try it. Thank you again!

JacopoPan, 2025-10-25:
Hi @wjh-scut 

for 1, [DSLPIDControl.py](https://github.com/utiasDSL/gym-pybullet-drones/tree/main/gym_pybullet_drones/control/DSLPIDControl.py) has `target_rpy_rates` although they are not currently used for external references, you'll need to modify `_preprocessAction()` in [BaseRLAviary.py](https://github.com/utiasDSL/gym-pybullet-drones/tree/main/gym_pybullet_drones/envs/BaseRLAviary.py)

for 2, just modify the `_observationSpace()` method in [BaseRLAviary.py](https://github.com/utiasDSL/gym-pybullet-drones/tree/main/gym_pybullet_drones/envs/BaseRLAviary.py)

wjh-scut, 2025-11-01:
@JacopoPan Hello, I noticed that `self.action_buffer` wasn't reset in the `reset` method. Could this be a problem? Because if it is, `self.action_buffer` will contain action information from the previous episode, which is incorrect for the new episode.

JacopoPan, 2025-11-02:
That's not a bad point but I'm not sure how influent it would be: you could add something like the following in `BaseRLAviary`:

```py
    def reset(self, seed = None, options = None):
        if self.ACT_TYPE in [ActionType.RPM, ActionType.VEL]:
            size = 4
        elif self.ACT_TYPE==ActionType.PID:
            size = 3
        elif self.ACT_TYPE in [ActionType.ONE_D_RPM, ActionType.ONE_D_PID]:
            size = 1
        else:
            print("[ERROR] in BaseRLAviary.reset()")
            exit()
        for i in range(self.ACTION_BUFFER_SIZE):
            self.action_buffer.append(np.zeros((self.NUM_DRONES,size)))
        return super().reset(seed, options)
```

and test using the `learn.py` examples whether the learning is faster/better behaved, I just added a few lines to plot out the episode reward vs. training steps with this commit d0aac0059a03b92c75713b073f2a10352de1476d