learnsyslab/gym-pybullet-drones #202: Ctrl Freq and Simulation Freq Questions

zcase, 2024-04-09:
@JacopoPan  in a previous issue thread you said:

>Ctrl freq is both the frequency at which observations are produced and actions are taken by the environment.
(Sim freq is the frequency at which the PyBullet step is called, normally greater than ctrl freq).

A lot of the default values are set at 30. Is this 30 in hertz? Also is sim frequency in hertz?

To your knowledge is there a case were you would want ctrl_freq > sim_freq?

JacopoPan, 2024-04-10:
The default sim/PyBullet frequency is 240Hz, the control frequencies in the examples are between 30 and 240Hz.
For low-level control, one might want to increase both for better numerical accuracy but making the control frequency greater than the simulation frequency will result in inputs/controls/actions being computed more often than the simulation state is updated which is unlikely to be useful.

ch-MEIJIE, 2024-06-19:
Hi,

Is it possible to set a lower control frequency, such as 2 Hz? I am currently working on an imitation learning task and need to collect demonstrations from users using the repository firstly. In my mind, I want to map some keys on the keyboard to discrete velocity values and user can choose one of them to press to change to drone's fly state. Considering the response time of humans, I believe setting the control frequency to 2 Hz is reasonable.

However, when I arbitrarily changed the value of DEFAULT_CONTROL_FREQ_HZ to 2 and ran pid_velocity.py, all drones fell to the ground.

I look forward to your reply. Thank you.