learnsyslab/gym-pybullet-drones #203: Clarification on Each Dimension's Meaning for ActionType.VEL

input-0313, 2024-04-14:
@JacopoPan 
Hello, 
I'm currently working on a project using the gym-pybullet-drones environment, and I'm implementing drone control using the ActionType.VEL. I observed that the action space is defined with a size of 4 for each drone when this action type is used.
Could you clarify what each of the four dimensions in this action vector represents specifically for ActionType.VEL? 

Additionally, in the "_preprocessAction" function, the target velocity vector is determined by the formula:
target_vel = self.SPEED_LIMIT * np.abs(target[3]) * v_unit_vector
Could you please explain why only the 4th dimension of the action vector is considered for the velocity's magnitude? Does this mean that the first three dimensions are intended for direction, and the fourth for scaling the velocity to the desired magnitude?

I'm trying to understand the underlying logic to ensure correct implementation and possibly make necessary adjustments for my project. 
Thank you for your help!

JacopoPan, 2024-04-15:
Hi @input-0313 

yes, you are correct, the first 3 components of the 4D action space in `ActionType.VEL` are interpreted as a direction in the 3D space, normalized to a unit vector, and then multiplied by the 4th component, interpreted as the desired velocity magnitude.

Of course, as you noted and if you prefer, you can modify this simply by re-implementing method `_preprocessAction` (part of the reason for doing this was simply to preserve the action space dimension across aviaries and not not change network architectures between learning example).

input-0313, 2024-04-16:
@JacopoPan Hello
Thanks for your previous reply. I'm implementing horizontal flight with the CF2X drone model, where the URDF file specifies a maximum flight speed of 30km/h. I've set the action space's xyz velocity thresholds at 8m/s before passing them into the controller, based on the speed unit in BaseRLAviary's "self.SPEED_LIMIT". 

However, the actual drone speeds seem substantially lower than these set points. Does PyBullet have an intrinsic speed limit, or are there constraints on the velocities fed into the controller? Additionally,  the drone occasionally flips or exhibits instability—might this be due to a need for PID tuning? If so, could you recommend some resources for PID parameter optimization?
Thank you for your time and assistance.

JacopoPan, 2024-04-17:
The control input (as individual motor RPMs) is computed from the desired velocity via the [DSLPIDControl](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/control/DSLPIDControl.py) class in these lines. The resulting desired thrust will be limited by the fact that the target position is the same as the current one, you could try to increase the `D_COEFF_FOR` coefficients in the controller (the tracking of the desired velocity will be more aggressive but the instability might be exacerbated).

https://github.com/utiasDSL/gym-pybullet-drones/blob/50d1a958def27121bcf63f3cc753049fee8e71e5/gym_pybullet_drones/envs/VelocityAviary.py#L158-L166