learnsyslab/gym-pybullet-drones #218: Sim2real transfer for betaflight

saghavart, 2024-05-24:
Hi. I am planning to use your framework for my Reinforcement Learning task. I saw that there is Betaflight support. My question is how can I achieve sim2real transfer for betaflight. Shall I write a new flight mode and build the new firmware? And also betaflight is using his own controller, so how is my RL code integrated on top of it? Are ther any succesful examples?

JacopoPan, 2024-06-02:
The instructions to run the example using Betaflight (in simulation) are detailed [here](https://github.com/utiasDSL/gym-pybullet-drones?tab=readme-ov-file#betaflight-sitl-example-ubuntu-only). It leverages the SITL version of BF to compute motor controls from CTBR commands.

The simulation environment in `gym_pybullet_drones/envs/BetaAviary.py` has a CTBR action space so, if you train an RL policy on it, it should output actions already compatible with BF. However, for sim2real transfer you also need to know the state of the system/drone. Are you going to use a MoCap system?

saghavart, 2024-06-03:
Thank you for answer. No, I want to deploy it outdoors. The only state I need are IMU,barometer and output from camera