learnsyslab/gym-pybullet-drones #220: Request to Migrate from Master to Main Branch for Improved Stability

abdul-mannan-khan, 2024-06-22:
Hi @JacopoPan 
Thank you for your dedication to maintaining this project. I have noticed several issues with the master branch that seem to limit its functionality.

Could we consider transitioning the primary development and examples from the master to the main branch to enhance stability and clarity for new contributors? I believe this could help avoid the current issues and streamline future development.

I am  eager to assist with this process to ensure a smooth transition. I just need your kind guidance for the issues I face in this process. Thanks for considering this change.

JacopoPan, 2024-06-22:
Hello @abdul-mannan-khan 
`main` is the current default branch of the repo, `master` is the original branch of this work (published in the form of branch `paper`), updated up until last year.

Because of the major changes in at least 3 of the main dependencies of this work (rllib, gym, stable-baselines3), I switched to maintaining `main` instead of `master` roughly one year ago: if you have made improvements/bugfixes to `master` I'm happy to include them (even if I don't use that branch anymore), my suggestion is to (1) fork it, (2) apply your corrections, (3) open a PR from your fork to `master` of this repo. I will review and merge those changes as appropriate.

abdul-mannan-khan, 2024-06-25:
Regarding `FlyThruGate`, I am getting following error 

```
Traceback (most recent call last):
  File "/home/yf23868/brl_ws/gym-pybullet-drones_20240620/experiments/learning/singleagent_20240625a.py", line 209, in <module>
    run(**vars(ARGS))
  File "/home/yf23868/brl_ws/gym-pybullet-drones_20240620/experiments/learning/singleagent_20240625a.py", line 137, in run
    eval_env = gym.make(env_name,
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 740, in make
    env_spec = _find_spec(id)
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 537, in _find_spec
    _check_version_exists(ns, name, version)
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 403, in _check_version_exists
    _check_name_exists(ns, name)
  File "/home/yf23868/anaconda3/envs/drones/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 380, in _check_name_exists
    raise error.NameNotFound(
gymnasium.error.NameNotFound: Environment `flythrugate-aviary` doesn't exist. Did you mean: `ctrl-aviary`?
```
 My flythrugate-aviary follows your same style as in the paper hover/flythrugate-aviary here. 
[FlyThruGateAviary.zip](https://github.com/user-attachments/files/15980029/FlyThruGateAviary.zip)
 and my flythrugate simulation file is [singleagent_20240625a.zip](https://github.com/user-attachments/files/15980040/singleagent_20240625a.zip). 

Can you please help me run here? 

Just to mention that  `__init__.py` in `gym_pybullet_drones/gym_pybullet_drones/envs` has FlyThruGateAviary defined. 

```
from gym_pybullet_drones.envs.BetaAviary import BetaAviary
from gym_pybullet_drones.envs.CtrlAviary import CtrlAviary
from gym_pybullet_drones.envs.HoverAviary import HoverAviary
from gym_pybullet_drones.envs.FlyThruGateAviary import FlyThruGateAviary
from gym_pybullet_drones.envs.MultiHoverAviary import MultiHoverAviary
from gym_pybullet_drones.envs.VelocityAviary import VelocityAviary
```
Thank you so much for your help.

abdul-mannan-khan, 2024-06-26:
I figured this out. I updated gym_pybullet_drones->`__init__.py` and updated the following part 

```
register(
    id='flythrugate-aviary-v0',
    entry_point='gym_pybullet_drones.envs:FlyThruGateAviary',
)
```

The complete code is as follows:

```
from gymnasium.envs.registration import register

register(
    id='ctrl-aviary-v0',
    entry_point='gym_pybullet_drones.envs:CtrlAviary',
)

register(
    id='velocity-aviary-v0',
    entry_point='gym_pybullet_drones.envs:VelocityAviary',
)

register(
    id='hover-aviary-v0',
    entry_point='gym_pybullet_drones.envs:HoverAviary',
)

register(
    id='multihover-aviary-v0',
    entry_point='gym_pybullet_drones.envs:MultiHoverAviary',
)

register(
    id='flythrugate-aviary-v0',
    entry_point='gym_pybullet_drones.envs:FlyThruGateAviary',
) 
```