learnsyslab/gym-pybullet-drones #223: Motor Delay

zcase, 2024-06-28:
@JacopoPan do you know of a way to induce motor delay for a given time amount using the _pyhiics function or would we need to use the _dynamics function to do this?

JacopoPan, 2024-07-31:
I think the easiest way is to start implementing motor delay in the aviaries with RPM inputs and simulation stepped by PyBullet and simply by adding a buffer of actions that matches the delay measured in simulation steps (of course, in this case, the simulation integration step should be smaller than the delay we want to introduce).

zcase, 2024-08-02:
I think I tried doing a first order low pass filter on the actions based of the motor delay and the simulation freq within the step function and that seemed to help