learnsyslab/gym-pybullet-drones #225: one_d_rpm breaks with different masses for HoverAviary

KevinHan1209, 2024-06-28:
Hi,

I am running into an odd issue where if I change the mass by a small amount, say 0.05 kg, using the `pybullet.changeDynamics()` method in the `_housekeeping()` method, the training for `one_d_rpm` gets stuck at a very small reward and never improves. In some cases, during the rendering/evaluation it just oscillates above and below 1m. I have no idea why this is. It seems that `one_d_pid` does not have this issue and works well. I'm also slightly confused about the 5% parameter when updating RPMs in relation to the hover RPM. Does anyone have any experience with this?

abdul-mannan-khan, 2024-07-18:
Yes. There is a bug in this `HoverAviary.py`. It only works for the given example. You make small changes like mass, or action or even goal point, it will not work. Particularly, if you change goal point from (0,0,1) to (1,0,1), it will not work. I guess due to some problems in actions.

KevinHan1209, 2024-07-19:
I don't think it's the policy or the environment (reward function) that's the problem but rather the simulation settings itself. It doesn't make sense that a reward function would suddenly fail with marginal changes in mass. I'm suspecting the PyBullet physics implementation may be wrong. I saw that the thrust2weight ratio was also set as a constant in the urdf file, which doesn't make sense to me either.

abdul-mannan-khan, 2024-07-19:
I agree with you. I am also facing the same problem. Given the fact that the developers are busy, I think it would be much better to make own environment using ChatGPT.

abdul-mannan-khan, 2024-07-23:
Any luck, my drone is also stuck in one reward and not able to get out of it.

JacopoPan, 2024-07-31:
Two main clarifications:
- @abdul-mannan-khan the `one_d` action spaces limit actuation to only vertical thrust so, of course, cannot be used to move in x-y
- @KevinHan1209 if you want to change the physical properties of the drone, the way I structured the code, you should do it in the robot description file (.URDF). The issue you are facing is almost certainly created by the +-5% interval because it's around the `self.HOVER_RPM` calculated from the thrust necessary to beat gravity, calculated from the mass in the URDF. (the reason to limit the action space to that 10% of possible RPMs is because of course only a small part of the range 0-20000+ that the motors are capable of is actually useful for leveled flight)

Hope this helps, btw, if you are looking for an environment where to randomize mass and inertia, we created [safe-control-gym](https://github.com/utiasDSL/safe-control-gym) specifically for that.

zjn20030811, 2026-09-04:
I am preparing a focused fix for the runtime-mass variant of this issue. The one_d_rpm preprocessing currently uses the URDF-derived HOVER_RPM even after a caller changes a loaded body mass with PyBullet. I plan to compute a per-drone hover trim from the live mass for PyBullet-backed physics, preserve the existing value when no live body is available, and add deterministic tests covering unchanged and randomized mass cases. This keeps the default mapping unchanged while making domain-randomized mass settings physically consistent.