learnsyslab/gym-pybullet-drones #226: Create custom simulation environment to train UAV collision avoidance

Astik-2002, 2024-07-02:
Hello. I'm using gym-pybuller-drones as my simulation platform to design and test collision avoidance algorithms for agile drones. I would like to create a custom simulation environment filled with cluttered obstacles (both static and dynamic), as well as multiple drones (later on). Can someone please share how to design a simulation environment where obstacles can be spawned at different locations and orientations? Here are two examples of environment that i'd like to create here (https://github.com/USTC-AIS-Lab/PE-Planner/blob/master/gif/github_video5.gif) and (https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/master/files/icra20_3.gif).

abdul-mannan-khan, 2024-07-08:
You can do this by modify environments in the folder `env`. The obstacles you want to add as you mentioned can be done. For reference, I have attached the file which I have created. 
[FlyThruObstaclesAviary.zip](https://github.com/user-attachments/files/16125161/FlyThruObstaclesAviary.zip)

Also register that environment in `__init__.py` (outside `env` folder). The content that file should be like this. 

```
from gymnasium.envs.registration import register

register(
    id='ctrl-aviary-v0',
    entry_point='gym_pybullet_drones.envs:CtrlAviary',
)

register(
    id='velocity-aviary-v0',
    entry_point='gym_pybullet_drones.envs:VelocityAviary',
)

register(
    id='hover-aviary-v0',
    entry_point='gym_pybullet_drones.envs:HoverAviary',
)

register(
    id='multihover-aviary-v0',
    entry_point='gym_pybullet_drones.envs:MultiHoverAviary',
)

register(
    id='flythrugate-aviary-v0',
    entry_point='gym_pybullet_drones.envs:FlyThruGateAviary',
)

register(
    id='flythruobstacles-aviary-v0',
    entry_point='gym_pybullet_drones.envs:FlyThruObstaclesAviary',
)
```

One thing I would like to mention that RL-algorithms have not beaten so far `FAST Planner` algorithm. All it can do is to memorize the path like a blind man memorizes the path. I think. For detail, please go through [this paper](https://arxiv.org/pdf/2301.01755) by Davide Scaramuzza.

abdul-mannan-khan, 2024-07-09:
This is pretty standard as the same `FlyThruGateAviary` just add this into right a place and it would work. Let me know if it does not.

Astik-2002, 2024-07-09:
I don't have the code for Flythrugateaviary. I tried modifying the PID example in main branch, but got this error ``` [ERROR] invalid initial_xyzs in BaseAviary.__init__(), try initial_xyzs.reshape(NUM_DRONES,3)
[ERROR] invalid initial_rpys in BaseAviary.__init__(), try initial_rpys.reshape(NUM_DRONES,3)
Traceback (most recent call last):
  File "/home/astik/gym-pybullet-drones/gym_pybullet_drones/examples/fly_flythrough.py", line 201, in <module>
    run(**vars(ARGS))
  File "/home/astik/gym-pybullet-drones/gym_pybullet_drones/examples/fly_flythrough.py", line 115, in run
    env = FlyThruObstaclesAviary(drone_model=drone,
  File "/home/astik/anaconda3/envs/drones/lib/python3.10/site-packages/gym_pybullet_drones/envs/FlyThruObstaclesAviary.py", line 32, in __init__
    super().__init__(drone_model=drone_model,
  File "/home/astik/anaconda3/envs/drones/lib/python3.10/site-packages/gym_pybullet_drones/envs/BaseRLAviary.py", line 79, in __init__
    super().__init__(drone_model=drone_model,
  File "/home/astik/anaconda3/envs/drones/lib/python3.10/site-packages/gym_pybullet_drones/envs/BaseAviary.py", line 212, in __init__
    self._housekeeping()
  File "/home/astik/anaconda3/envs/drones/lib/python3.10/site-packages/gym_pybullet_drones/envs/BaseAviary.py", line 486, in _housekeeping
    self.DRONE_IDS = np.array([p.loadURDF(pkg_resources.resource_filename('gym_pybullet_drones', 'assets/'+self.URDF),
  File "/home/astik/anaconda3/envs/drones/lib/python3.10/site-packages/gym_pybullet_drones/envs/BaseAviary.py", line 487, in <listcomp>
    self.INIT_XYZS[i,:],
AttributeError: 'FlyThruObstaclesAviary' object has no attribute 'INIT_XYZS'
numActiveThreads = 0
stopping threads
Thread with taskId 0 exiting
destroy semaphore
semaphore destroyed
Thread TERMINATED
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed ```

abdul-mannan-khan, 2024-07-09:
FlyThruGateAviary is available in the `master` branch. By the way, it would be a lot easier to start with `master` branch using 

`git clone -b master https://github.com/utiasDSL/gym-pybullet-drones.git`

just modify the following line in the 

`from stable_baselines3.common.cmd_util import make_vec_env`

to this 

`from stable_baselines3.common.env_util import make_vec_env`

and it should work flawlessly. 

I did nothing special. Just copy the file I sent you and paste it where all other files are `env` files are placed, modify the `__init__.py` file as I mentioned previously and add the environment using 

`from gym_pybullet_drones.envs.FlyThruObstaclesAviary import FlyThruObstaclesAviary`

Thanks.

Astik-2002, 2024-07-09:
Did you mean `master` branch or `main` branch? In `master` branch, BaseAviary and BaseRLAviary are not available. I'm trying in masters rn, and got this issue.

Astik-2002, 2024-07-10:
Its working now, thanks.

closed 2024-07-10 by Astik-2002

Astik-2002, 2024-09-04:
@INDEX108 you need to create a urdf file for a cylinder and place it in the assets folder. You can ask chatgpt to create one for you

INDEX108, 2024-09-04:
Thank you for your reply. I've temporarily replaced it with box. urdf and it works. But why do obstacles always fall so fast that they can't stay in place?

abdul-mannan-khan, 2024-09-04:
There are few things to play with to solve this problem: Inertia, mass etc

AI4IS, 2024-09-28:
> There are few things to play with to solve this problem: Inertia, mass etc

Hey, it seems that you haven't change observation space, and it may lead to a bad result cuz the NN network can't build a map from meaningless observation to action.