learnsyslab/gym-pybullet-drones #233: Deviations in the actual CTRL_FREQ for the training loop

Chicean, 2024-07-30:
Hi, @JacopoPan 
I find out that there is a `sync()` method to control the `env.step()` frequency (namely the `CTRL_FREQ` parameter) in the testing env loop, but the training env does not contain such a mechanism, which may lead to deviations in the actual `CTRL_FREQ`. Is such a design reasonable? Will deviations in the actual `CTRL_FREQ` affect the final performance of the drone after RL learning? Any insights will be highly appreciated!!
https://github.com/utiasDSL/gym-pybullet-drones/blob/3d7b12edd4915a27e6cec9f2c0eb4b5479f7735e/gym_pybullet_drones/examples/learn.py#L182

JacopoPan, 2024-07-31:
Hi @Chicean 

the `sync` method is really only just a sleep call to slow down the simulation to a clock closer to the wall clock when visualizing through the GUI, in controls and rl experiments, when simulation stepping and control input are in lockstep, this shouldn't matter.

Chicean, 2024-07-31:
Hi, @JacopoPan
Thank you so much for the detailed explanation! ! I now have a much deeper comprehension of `sync()` method and `CTRL_FREQ` parameter.

closed 2024-07-31 by Chicean