learnsyslab/gym-pybullet-drones #237: Increase the range of depth camera

Astik-2002, 2024-08-16:
Greetings. Is it possible to increase the max range of depth camera. Camera's like intel realsense d455 provide accurate depth to upto 4 meters. I was hoping to simulate the same. I using open3d to convert the depth image to a point cloud and publish it to ros, but so far the range I'm getting is 1m in front of the camera.

https://github.com/user-attachments/assets/283b1302-b437-45eb-a9e9-90306e84bf80

JacopoPan, 2024-10-10:
Hi @Astik-2002 

the implementation do the depth buffer in this repo is simply taken from PyBullet, if you need to customize/better understand it, you should refer to
- page 51 of this [manual](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit)
- https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/pointCloudFromCameraImage.py
- https://stackoverflow.com/questions/6652253/getting-the-true-z-value-from-the-depth-buffer

Astik-2002, 2024-10-10:
Hi @JacopoPan. Thanks for the reply. I've been able to fix the range problem by de-normalizing the depth values with the nearval and farval parameters, that opengl uses. It allowed me to create proper pointclouds. Once I've figured out the path following stuff I'd try and make a pr for this