learnsyslab/gym-pybullet-drones #243: Implementing a Custom Controller on Gym-Pybullet Drones

lakshmikanth-nageswar, 2024-10-28:
Hi, could anyone please clear these queries of mine:

- How to modify the training tasks say for hover, single set-point (regularisation) and tracking problem through this framework? and for visualisation of trained policy in pybullet as well. (how to modify the BaseAviary or VelocityAviary for tracking and other scripts required to be modified?)

- How to implement our own controller (say linear or non-linear or learning based) in this environment

Thanks in Advance.

JacopoPan, 2024-10-30:
I'm not sure I understood your first questions (most of those things are already implemented, pybullet is the api towards the physics/rendering backend, the learning is using stable-baselines3 which is PyTorch based).

For the second one, you can subclass [BaseControl.py](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/control/BaseControl.py) as in, for example, [DSLPIDControl.py](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/control/DSLPIDControl.py)

lakshmikanth-nageswar, 2024-10-30:
Thanks for the help!

closed 2024-10-30 by lakshmikanth-nageswar