learnsyslab/gym-pybullet-drones #244: Controlling Crazyflie through RPMs (PWM)

piratax007, 2024-11-11:
Hello, I have successfully trained a policy for setpoint tracking using RPM as action space. Now I'm trying to deploy it to a real Crazyflie using a ROS2 package (Crazywarm and Vicon system). As you refer to in #118, exists an empirical study of the relation between PWM and RPMs for Crazyflie (https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/pwm-to-thrust/). My question, is that using the data from that study and comparing with the transformation used by GymPybullet I found high discrepancies as is shown in the plot. The black line corresponds with the linear transformation used by GymPybyllet.
[image]
Could you help me to understand this and how can affect the transferability of the trained policy to a real drone?

Thank you so much

JacopoPan, 2024-11-26:
The relation between RPMs and thrust in `gym-pybullet-drones` is modeled as quadratic:
https://github.com/utiasDSL/gym-pybullet-drones/blob/3d7b12edd4915a27e6cec9f2c0eb4b5479f7735e/gym_pybullet_drones/envs/BaseAviary.py#L693-L694

There assumption of linearly between RPMs and PWM (that don't fully hold away from a hover range) in `DSLPIDControl.py` and `CFAviary.py` but those shouldn't come into play if you purely trained RL for RPMs in a class derived from BaseRLAviary.py

piratax007, 2024-12-13:
I understand that training using RPMs as actions space the conversion to PWM doesn't play a role. Nevertheless, transferring a policy trained to produce RPMs to a real Crazyflie, requires translating the RPM to PWM because the lowest control signal received by the Crazyflie firmware is PWM.

`BaseRLAviary` translates the policy outputs to RPM through
```
target = action[k, :]
if self.ACT_TYPE == ActionType.RPM:
    rpm[k,:] = np.array(self.HOVER_RPM * (1+0.05*target))
```

Then, to obtain the PWM should be applied:

$PWM = \dfrac{RPM - 4070.3}{0.2685}$

Is it correct?

JacopoPan, 2024-12-15:
In `BaseAviary`, the relationship between RPMs and thrust is a simple quadratic one using coefficient `kf="3.16e-10"` which is taken from the system identification resources lined in the .urdf
```
<!-- links>
      <carlos url="https://arxiv.org/pdf/1608.05786.pdf" />
      <julian url="https://www.research-collection.ethz.ch/handle/20.500.11850/214143" />
      <mit url="http://groups.csail.mit.edu/robotics-center/public_papers/Landry15.pdf" />
</links -->
```

If you want to go by the system identification in https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/pwm-to-thrust/ you should modify
https://github.com/utiasDSL/gym-pybullet-drones/blob/3d7b12edd4915a27e6cec9f2c0eb4b5479f7735e/gym_pybullet_drones/envs/BaseAviary.py#L693
so that thrust = 1.0942e-07rpm^2 - 2.1059e-04rpm + 1.5417e-01 instead
and then consider that thrust = 0.409e-03rpm^2 - 140.5e-03rpm - 0.099 when you want to derive the corresponding PWM command