learnsyslab/gym-pybullet-drones #256: Questions about the crazyflie model and simulation

xchl19, 2024-12-27:
Hello,

Thanks for your great work of developing the quadrotor simulatior.

I have several questions about the crazyflie model and simulation.

1. When I check the figure of thr drone in paper and official site of bitcraze, I noticed that the order of motors are different (in the reverse order). When checking the `_physics()` in BaseAviary.py the sign of z_torque is different from paper https://arxiv.org/pdf/1608.05786, I wonder if the rotation direction of motors is different from the official website?

2. I am not familar with the Euler angles in pybullet. I wonder in this simulator, what is the order of rotations axes, is it intrinsic or extrinsic rotation?

3. When I use `dyn` as physics, the drones fell down to underground. Are there bugs in the dynamic?

JacopoPan, 2025-01-04:
Hello @xchl19,

1. the implementation here is mainly taken by the system identification works mentioned in the URDF files
```
<carlos url="https://arxiv.org/pdf/1608.05786.pdf" />
<julian url="https://www.research-collection.ethz.ch/handle/20.500.11850/214143" />
<mit url="http://groups.csail.mit.edu/robotics-center/public_papers/Landry15.pdf" />
```
not Bitcraze documentation directly, if you are using a specific firmware, you could indeed have to remap the inputs as it is done in `BetaAviary` and `CFAviary`
https://github.com/utiasDSL/gym-pybullet-drones/blob/3d7b12edd4915a27e6cec9f2c0eb4b5479f7735e/gym_pybullet_drones/envs/BetaAviary.py#L258
https://github.com/utiasDSL/gym-pybullet-drones/blob/3d7b12edd4915a27e6cec9f2c0eb4b5479f7735e/gym_pybullet_drones/envs/CFAviary.py#L633

2. see the ["Transforms" section in PyBullet Quick Guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit?tab=t.0#heading=h.y6v0hy52u8fg) for the rotation order

3. running which one of the example?

xchl19, 2025-01-05:
Thanks a lot for your reply!

About question 3, I changed `DEFAULT_PHYSICS = Physics("pyb")` in `pid.py` into `DEFAULT_PHYSICS = Physics("dyn")`.

JacopoPan, 2025-01-05:
Thanks @xchl19, it seems you are correct and "dyn" is not working as expected
[This](https://github.com/utiasDSL/gym-pybullet-drones/pull/166) was the most recent change to its implementation but even reverting it does not seem to fix the issue, I'll need to investigate a bit deeper if something about the robot description or bullet has changed

JacopoPan, 2025-08-30:
Thanks for pointing that out, @Shenzhaolong1330 , option `dyn` should now work even without changing mixer matrix

closed 2025-09-18 by JacopoPan