learnsyslab/gym-pybullet-drones #259: Possible noise model for depth image

Astik-2002, 2025-01-07:
@JacopoPan is it possible to add some kind of a sensor model, that adds noise to the depth image? I found this repo https://github.com/ankurhanda/simkinect where noise is added to a rendered depth map, but the code requires disparity computation, which is hard to do as in pybullet we just have a single depth image (unlike a stereo camera with two rgb images, separated by a distance). Can you share if there's any other way to add noise to the image so that it's more alike to depth image from cameras like zed2 or intel realsense?

Astik-2002, 2025-01-07:
Also, can you share the limits of acceleration/deceleration for the base crazyflie model used in the simulation?

Astik-2002, 2025-01-13:
@JacopoPan Thanks for the assist. 

Can you share if there's a way to include the acceleration of the UAV in the observation state vector? Is there a simulated imu which I can access for values?

JacopoPan, 2025-01-14:
Unfortunately not, artificial sensors are not implemented here, but you might attempt to differentiate the velocities to approximate a rough estimate