learnsyslab/gym-pybullet-drones #282: Calculating rewards for multyagent task

Ange1ika, 2025-09-16:
Hello! I want to learn how to control a swarm of UAVs with obstacles. Could you tell me, please, in what order to change the code? Which action to use and how to work with it correctly. How to figure out the rgb calculation?

I'm figuring out the pipeline, I found adding obstacles, but there are difficulties with calculating rewards. Can you explain how you got these threshold values? Why such a formula

`
    #### Target cumulative rewards (problem-dependent) ##########
    if DEFAULT_ACT == ActionType.ONE_D_RPM:
        target_reward = 474.15 if not multiagent else 949.5
    else:
        target_reward = 467. if not multiagent else 920.
    callback_on_best = StopTrainingOnRewardThreshold(reward_threshold=target_reward,
                                                     verbose=1)
`

JacopoPan, 2025-09-19:
Hi @Ange1ika 

a few comments:
- to create your problem-specific learning environment, you need to subclass [BaseRLAviary]https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/envs/BaseRLAviary.py) 
- [MultiHoverAviary](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/envs/MultiHoverAviary.py) is one such example of a `BaseRLAviary` subclass 
- The reward function is then computed in this subclass by method `_computeReward()`
- The `ActionType` that are prefixed by `ONE_D` only allow movement on the z-axis, you will want to use `PID` or `VEL`

Ange1ika, 2025-09-19:
Thank you! I chose pid control now. 

Am I correct in understanding that multi-agent learning requires using RGB space (taking obstacles into account). Then, the distance to objects and reward information can be obtained via RGB? Can I see some examples? Here, if I've understood correctly, it's only without obstacles:

gym-pybullet-drones/experiments/learning
/multiagent.py

Ange1ika, 2025-09-19:
> Hi [@Ange1ika](https://github.com/Ange1ika)
> 
> a few comments:
> 
> * to create your problem-specific learning environment, you need to subclass [BaseRLAviary](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/envs/BaseRLAviary.py)
> * [MultiHoverAviary](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/envs/MultiHoverAviary.py) is one such example of a `BaseRLAviary` subclass
> * The reward function is then computed in this subclass by method `_computeReward()`
> * The `ActionType`s that are prefixed by `ONE_D_` only allow movement on the z-axis, you will want to use `PID` or `VEL`

Thank you! I chose pid control now.

Am I correct in understanding that multi-agent learning requires using RGB space (taking obstacles into account). Then, the distance to objects and reward information can be obtained via RGB? Can I see some examples? Here, if I've understood correctly, it's only without obstacles:

gym-pybullet-drones/experiments/learning
/multiagent.py

Ange1ika, 2025-09-19:
[image]

I'm currently using MLP, but the drones freeze at times. I need more time; RGB observations are better handled by CNN.

And I'm still thinking about what policy to follow; it would be good to have different actors and one critic, as in your article.

JacopoPan, 2025-09-21:
> > I’d definitely recommend expanding the `KIN` observation space to account for drone positions, using `RBG` with multiple vehicles would be computationally very wasteful.
> 
> But with kin I won't be able to add obstacles or will I?

You could simply extend the observation with position and dimension of the obstacles

Ange1ika, 2025-09-21:
> You could simply extend the observation with position and dimension of the obstacles

Okey, I understood. Thank you very much!!