learnsyslab/gym-pybullet-drones #283: Obstacle avoidance and Navigation

eOvic, 2025-09-28:
I was wondering if there is a file or an example of someone using this framework for navigation, I made a custom environment which is a sub class of BaseRLAviary and When I train the PPO or TD3 it takes a lot of time to go from point a to b. I tried curriculum learning but still the convergence is way too far and the learning take a lot of time.

JacopoPan, 2025-09-29:
I’m sure a Gymnasium compatible wrapper can be made even around gz sim/px4 sitl (although it might be a bit clunkier) and I have it in my todo list

JacopoPan, 2025-09-29:
Even if it adds a bit of overhead, I would restart px4 sitl between episodes (possibly even gazebo). One advantage of the containerized approach I put together is that would allow to parallelize this without too much pain (to offset the other computational overheads). (If you eventually wanted to contribute it as a PR, I'd be happy to test/review)

JacopoPan, 2025-12-09:
See: https://github.com/JacopoPan/aerial-autonomy-stack?tab=readme-ov-file#gymnasium-environment